yy502/inkscape-laserpecker #1: Interior engraving

miltonardila, 2020-07-13:
Hi, I downloaded and installed your plugin for Inkscape. I have created an SVG with the text I want to engrave in wood, but when I create the G-CODE and visualize it in NC, just engrave the outline.
I'm enclosing the SVG and the G-CODE if you can help me, thank you.

[Archivo comprimido.zip](https://github.com/yy502/inkscape-laserpecker/files/4913342/Archivo.comprimido.zip)

yy502, 2020-07-13:
I have just come this issue myself yesterday! For a long time, I accepted that Gcode can only do the outline, but in fact you can fill your pattern with lines!

https://github.com/KnoxMakers/KM-Laser

Install this extension, before generating gcode with my extension, fill the shape following this tutorial :D 

https://www.youtube.com/watch?v=qdIjZXzT-QE

miltonardila, 2020-07-14:
> https://github.com/KnoxMakers/KM-Laser

Thank you for your recommendation. I downloaded this extension and used it. But I don't know how to control a more suitable filling. I've attached a picture so you can see what I'm talking about.

As you can see, I think this plotting makes it very slow. The final image is 70 x 50 mm, laser speed 500 and laser power 255. 2 passes and it took about 30 minutes in total, that's a long time. I don't know if images like this are better by bitmap, so that the engraving is linear and doesn't go around so much.

With the parameters of your extension "laser speed" will be the same "depth", right?

_**The final question is, how do you think I can improve the engraving with your G-CODE extension and KM-Laser padding for images of this type?**_

[image]

[image]

[image]

yy502, 2020-07-14:
Hi there.

Yes, setting speed is same as setting depth. deeper = slower.

If you want 100% fill, I think Bin mode in the App should work just fine. Gcode will make the laser travel longer paths for the same result.

To save some time by reducing fill percentage, you can use parallel line fill or cross line fill, with larger spacing.

Your job there used the default fill mode of the extension, which is Crosshatch.
If you uncheck it, you will get one-way parallel line fill, which is faster (half the travel distance hence half the time).

Note that you can change both the hatch spacing and angle.
I've attached a sample file. I suggest you reduce your shape's stroke thickness so you can the paths clearly.

If you only need one-way parallel line fill, I suggest you give the app's built-in gcode mode a try. It's better than I thought. 

Hope this helps.

[hatch_fill_example.svg.zip](https://github.com/yy502/inkscape-laserpecker/files/4920433/hatch_fill_example.svg.zip)

miltonardila, 2020-07-15:
Thank you very much for your comments and help.

yy502, 2020-07-15:
happy to help. I've updated the readme to include some instructions on how to use KM Laser extension.

closed 2020-07-15 by yy502