yy502/inkscape-laserpecker #2: Speed Setting in G-Code file ignored by Laserpecker L1

christianroze, 2020-10-05:
Hello, the speed setting in G-Code file will be ignored by Laserpecker. Pecker always travels with 3000mm/min and Laser Power stays at 1%.  If for example travel speed 200mm/min is selected your code produces G1 F200, that will be ignored by my Laserpecker.
If i append ".000000" (G1 F200.000000) it works great.
Laser Pecker App is 3.45
Firmware of Laserpeckerpecker ist 1.02

yy502, 2020-10-05:
can you send me your sample gcode file for a test, please?

christianroze, 2020-10-05:
[rectangle_circle.txt](https://github.com/yy502/inkscape-laserpecker/files/5327537/rectangle_circle.txt)

yy502, 2020-10-05:
Thanks for the sample. it worked for me as-is, but then my L1 has firmware v1.0. It could be a bug in firmware v1.02.

I'll add your workaround to the extension as it does not conflict with anything else.

yy502, 2020-10-05:
Updated. Could you have a try and let me know if that fixes your issue?

christianroze, 2020-10-06:
Cutting speed works now, but travel Speed (move fast) is ignored, stays in cutting speed.
First ;move fast is G1 F3000.0 seems to be OK
but then next 
;move fast
G1 F3000.0.0

yy502, 2020-10-06:
silly typo... fixed

christianroze, 2020-10-06:
Now it works very well. Thank you for the nice Inkscape Plugin.

closed 2020-10-06 by christianroze