yy502/inkscape-laserpecker #4: Error code when running Oversized Gcode Generator for L1

goodwinr, 2021-01-19:
Can you please help. 

 File "laserpecker_l1_oversize.py", line 54
    def bezierslopeatt(((bx0,by0),(bx1,by1),(bx2,by2),(bx3,by3)),t):
                       ^
SyntaxError: invalid syntax

Do you have a Good gcode file I can try.

Regards Ronnie

yy502, 2021-01-19:
Hi there, this is experimental and only worked with Inkscape 0.9 and L1 (firmware v1.0) specifically.

I don't recommend using it as it might damage your engraver.

I have no plan to fix it.

closed 2021-01-19 by yy502

goodwinr, 2021-01-19:
Hi, Thanks for the quick reply, if there is a possibility of damaging the engraver I will not use it. RegardsRonnie-------- Original message --------From: Yi Yu <notifications@github.com>Date: Tue, 19 Jan 2021, 18:44To: yy502/inkscape-laserpecker <inkscape-laserpecker@noreply.github.com>Cc: goodwinr <ronniegoodwin@supanet.com>, Author <author@noreply.github.com>Subject: Re: [yy502/inkscape-laserpecker] Error code when running Oversized Gcode Generator for L1 (#4)
Hi there, this is experimental and only worked with Inkscape 0.9 and L1 (firmware v1.0) specifically.
I don't recommend using it as it might damage your engraver.
I have no plan to fix it.