yy502/inkscape-laserpecker #5: "IndexError: list index out of range" when trying to generate G-Code for imported svg

croeer, 2021-02-19:
Hi there, I traced a little picture and opened in inkscape. When trying to generate the g-code i get the following error message:
```
Traceback (most recent call last):
  File "laserpecker.py", line 1470, in <module>
    e.run()
  File "C:\Users\croeer\Downloads\inkscape\share\inkscape\extensions\inkex\base.py", line 140, in run
    self.save_raw(self.effect())
  File "laserpecker.py", line 1451, in effect
    self.orientation(self.layers[min(0, len(self.layers) - 1)])
  File "laserpecker.py", line 1386, in orientation
    doc_height = self.svg.unittouu(self.document.getroot().xpath('@height', namespaces=inkex.NSS)[0])
IndexError: list index out of range
```
Example file: [toad.zip](https://github.com/yy502/inkscape-laserpecker/files/6008795/toad.zip)

I was able to fix it using a deprecated function call in `laserpecker.py,` so i could generate the g-code, but maybe you can look into it:
```
        doc_height = self.svg.unittouu(self.getDocumentHeight())
```
I'm also new to inkscape, so maybe I am doing something else wrong.

yy502, 2021-02-19:
Thanks for the sample file and workaround. I'll look into this over the weekend.

yy502, 2021-02-20:
Hi there, I did a quick test. Whatever shape/path I created in your svg file will cause this error. If I copy & paste your toad to a new Inkscape doc, it will work fine. So I guess it's something to do with the XML attributes of your svg file.

[toad.txt](https://github.com/yy502/inkscape-laserpecker/files/6013914/toad.txt)

croeer, 2021-02-20:
OK I see the problem: I created the svg file with an app called Vectornator, that has a "responsive" toggle on svg export. If unset, no width/height attributes are written so the svg element, which leads to the index error. 
If set, height is set to 100%, which then messes with the units, i.e. even if set to mm the extension will still complain about the image being to big, so I have to rescale it to be less than 100x100 px (!) (also, some weird path errors occur, where your comments of ungroup, object to path helped). 

I did however find an undeprecated way to get the height: use ```svg.height```:
```
doc_height = self.svg.unittouu(self.svg.height)
```

The generated g-code is off by 5x10E-5 mm when compared to the original version. Perhaps this is a fix you could incorporate, so the extension supports responsive svg files not created directly from inkscape.

Thanks for your quick support, the great extension and your overall helpiness on fb.

closed 2021-11-11 by yy502