yy502/inkscape-laserpecker #6: Is automatic engraving Gcode via USB possible?

PastuDan, 2021-11-02:
Thanks for the detailed documentation!

I just purchased a laserpecker, and would like to use this device to engrave serial numbers on a batch of items, and other stats specific to each item.

For this, I definitely think I can use your code or modify it slightly to automatically generate the gcode files from the SVGs / images that I generate.

However, I'm curious if its possible to get the generated gcode over to the device without using the Laserpecker Windows or OSX software?

Thanks!

yy502, 2021-11-02:
I'm afraid not. I've been chasing LP for an API, but they are not keen on developing it. So we are limited to using the LP mobile or PC app.

yy502, 2021-11-02:
For the L1 Pro, the USB interface is for power only. You can only control it via BLE protocol. But again, I got as far as connection and reading user email + safety pin from the engraver. The engraving commands and image encoding are unavailable.

PastuDan, 2021-11-02:
Bummer :-( I was about to ask if you thought it might be possible to read the USB packets using something like USBPcap / Wireshark. If its just a USB to serial bridge that sends gcode this might be pretty trivial.

But it sounds like it might not be possible at all -- oh well.

I'll dig into it a bit more when my laserpecker comes in, but thank's for the quick response!

yy502, 2021-11-02:
no problem. I've spent lots of time trying to reverse engineer it, so I can control it from my computer, but had not much luck.

you can use a BLE debugging app on your phone to connect to the LP Pro and play around.

On the computer, I was able to use `gatttool` to connect to it.

PastuDan, 2021-11-02:
Hey @yy502 would an Ubertooth One help reverse engineering it?

I'd be happy to send one your way if you'd like to take a stab at it.

yy502, 2021-11-02:
I appreciate your offer, but I don't think I need it.

I have an LP2, which can be controlled over USB serial. By looking at the handshake part, it's the same data format as the LP1 (Pro). So I can just sniff serial packets for reverse engineering when I have time.

As far as I was told that the engraving commands including data format are encrypted. Not sure if that's true or just a lie to stop me from trying 😆

PastuDan, 2021-11-02:
Ah, ok then I might be buying an LP2 as soon as I can! Thanks for all the help so far

closed 2021-11-02 by yy502

Cronic-se, 2024-04-07:
@pastudan 2,5 years ago. I am looking for the samt thing. Engraving for serial production. Did you solve this in some way? Another solution / product maybe?

icepicknz, 2024-04-19:
> @pastudan 2,5 years ago. I am looking for the samt thing. Engraving for serial production. Did you solve this in some way? Another solution / product maybe?

I too am trying to do the same thing, batch engraving QR codes. Let me know if you find a solution.

Apparently the android app can be setup for serialisation but its not on windows/Mac unfortunately.

This is the instructions they sent me that may help you and if you get it working I'd love to know as I dont have an android but happy to buy one if it will resolve my manual job

`Thank you for your reply. We regret to inform you that this function is not supported on the PC side currently， as it is still in the development stage. Please stay tuned for updates!

You can use the Slide Extension to achieve serially laser in serial numbers by turning on the Slide Extension multi-file engraving mode of Android LDS.
Note: Engraving materials need to be placed on the Slide Extension in advance. The serial number content that needs to be engraved also needs to be edited in the APP in advance.
 
This is the slide operation guide: https://www.youtube.com/watch?v=sC1EUqjHW70
 
Besides, this is a video of batch button box used with LP2. It is also used on LP4. For your reference: 
https://www.youtube.com/watch?v=a1ERtL2MRjE
 
Hope it helps. If you have any questions please feel free to contact us.😉️
`