yy502/inkscape-laserpecker #18: laserpecker.py:1145: SyntaxWarning: invalid escape sequence '\s'

rotragit, 2025-01-06:
Inkscape 1.2.2 (b0a8486541, 2022-12-01)
Ubuntu 24.04.1 LTS
Python 3.12.7
LP1 Pro

```
/home/romano/.config/inkscape/extensions/laserpecker.py:1145: SyntaxWarning: invalid escape sequence '\s'
  if type_.lower() in re.split("[\s\n,\.]+", errors.lower()):
/home/romano/.config/inkscape/extensions/laserpecker.py:1149: SyntaxWarning: invalid escape sequence '\s'
  elif type_.lower() in re.split("[\s\n,\.]+", warnings.lower()):
/home/romano/.config/inkscape/extensions/laserpecker.py:1153: SyntaxWarning: invalid escape sequence '\s'
  elif type_.lower() in re.split("[\s\n,\.]+", notes.lower()):
/home/romano/.config/inkscape/extensions/laserpecker.py:1274: SyntaxWarning: invalid escape sequence '\s'
  r = re.match("^\s*.\s*(\S+)", path.get("d"))
No path selected. Using all available paths.

Note that any shape not yet converted to path is omitted.
```

Output file is 0 byte.

yy502, 2025-01-08:
these are just some warnings, which shouldn't have caused any issues in Gcode generation.

`No path selected. Using all available paths.`

I suspect there was no path in your original design. Can you select all shapes and convert them to paths from the **Path** menu and try again? Otherwise, could you attach the SVG file that had this issue so I can take a look?

rotragit, 2025-01-08:
[image]

Well, it's just a circle drawn in Inkscape.

yy502, 2025-01-08:
I'm afraid I couldn't reproduce the issue with the file you supplied. I tested with Inkscape 1.2.2 on a Mac.
I can test again on a Ubuntu box when I get a chance.

[image]
[test_lp1pro_200_249.txt](https://github.com/user-attachments/files/18352458/test_lp1pro_200_249.txt)

rotragit, 2025-01-08:
I think it's related to the python version. Which python version have you
on your mac?
Thank you anyway.

yy502, 2025-01-08:
It's 3.10, shipped with Inkscape

`yi@yi-tm02580-mbp bin % pwd
/Applications/Inkscape.app/Contents/Frameworks/Python.framework/Versions/Current/bin
yi@yi-tm02580-mbp bin % ls -l
total 184
lrwxr-xr-x  1 yi  admin      9  3 Dec  2022 2to3 -> 2to3-3.10
-rwxr-xr-x@ 1 yi  admin    100  6 Oct  2022 2to3-3.10
lrwxr-xr-x  1 yi  admin      8  3 Dec  2022 idle3 -> idle3.10
-rwxr-xr-x@ 1 yi  admin     98  6 Oct  2022 idle3.10
lrwxr-xr-x  1 yi  admin      7  3 Dec  2022 pip3 -> pip3.10
-rwxr-xr-x@ 1 yi  admin    228  6 Oct  2022 pip3.10
lrwxr-xr-x  1 yi  admin      9  3 Dec  2022 pydoc3 -> pydoc3.10
-rwxr-xr-x@ 1 yi  admin     83  6 Oct  2022 pydoc3.10
lrwxr-xr-x  1 yi  admin     10  3 Dec  2022 python3 -> python3.10
lrwxr-xr-x  1 yi  admin     17  3 Dec  2022 python3-config -> python3.10-config
-rwxr-xr-x@ 1 yi  admin  70144  3 Dec  2022 python3.10
-rwxr-xr-x@ 1 yi  admin   2042  6 Oct  2022 python3.10-config`

rotragit, 2025-01-18:
Changing regex syntax from

`re.split("[\s\n,\.]+", errors.lower())`

to

`re.split(r'[\s\n,\.]+', errors.lower())`

(and the other lines in similar way) solve the problem.